worproject/RPi-Windows-Drivers #3: GPU Driver not available

SergiySeletsky, 2020-10-12:
Hi, it would be great to have GPU acelleration for Windows on RPi 3/4. Not Windows UI is too slow for any usage.
What is blocking finishing this driver? Any help needed?

sojab0on, 2020-11-07:
there was one but its old and not realy good to work with for normal use its a driver based on an sample render solution wich is not completely working.

after trying it my clean w10 crached and refuses to boot

jankais3r, 2020-11-09:
There was an announcement on the WoR Discord yesterday that GPU drivers are being worked on right now.

megapro17, 2020-12-11:
@jankais3r any progress?

sojab0on, 2020-12-11:
No not as far as i know

worproject, 2020-12-12:
I don't want to ruin anyone's dreams, but without implication from Microsoft, Broadcom / RPi Foundation, this has almost zero chances of happening in the near future.
The same goes for Wi-Fi.

sojab0on, 2020-12-13:
As far as on the discord its stated the base is there but will take more time due to no source availeble but they say they are still on it working

nguyenthienminh, 2022-06-28:
It's already 2 years already, is the 3D acceleration thing working on Windows 11 on Raspberry Pi 4 now?

sojab0on, 2022-06-28:
No , probably it will not in the near future as well, if it was there already it was announced already 

writing drivers from scratch is not an easy nor quick task to do, so please be patient

megapro17, 2022-06-28:
Please be patient? Project is mostly dead, even domain has been lost

sojab0on, 2022-06-28:
The changed it , there is a discord server where they told that. Will check new one if tou want it

binarymaster, 2023-07-20:
> I don't want to ruin anyone's dreams, but without implication from Microsoft, Broadcom / RPi Foundation, this has almost zero chances of happening in the near future.

Ampere guys got ARM WDDM driver for ASPEED AST2500 video: https://twitter.com/x86corez/status/1681391025175003144, so now there should be higher chances.

Just need a good documentation for the Broadcom VideoCore. The WDDM is already documented enough.

MinGW32-2006, 2023-07-22:
> Ampere guys got ARM WDDM driver for ASPEED AST2500 video: https://twitter.com/x86corez/status/1681391025175003144, so now there should be higher chances.
>
> Just need a good documentation for the [...]

However AST2500 seems only to have 2D acceleration support.
And I'm curious if there's anyone working on the gpu driver.