worproject/RPi-Windows-Drivers #7: Info on wifi driver

aiden1989acw, 2020-12-19:
Found this information on the following link that should assist with the WIFI drivers on the following link:

https://mail-index.netbsd.org/port-arm/2017/07/30/msg004328.html

Here is the context on the link.

"I've added a new driver for the Raspberry Pi's SD card controller to -current. The BCM283x family includes two SD card controllers; one is an Arasan SDHCI compliant controller (sdhc(4) driver) and another internal SD HOST controller (the new sdhost(4) driver).

Up until now, we have been using the SDHCI controller to drive the SD card slot. The SD HOST controller cannot do SDIO, so using it for the SD card slot will allow us to free up the SDHCI controller for use with the Wi-Fi chip on the Pi 3 and Zero W boards. "

Possibly changing the modifying the SD driver as above should assist with the SDIO interface and allow for WL_HOST_WAKE being triggered via the WLAN driver

Device manager does not show the Wifi portion of the CYW43445 as far as i can see, I only have the following listed:
1) ACPI\BCM2850\0
2) ACPI\BCM2854\0
3) ACPI\BCM2855\0
4) ACPI\PNP0C06\2&DABA3FF&0
5) ACPI\BCM2835\0

Does this require WLAN_ON request onboot via the BIOS? 

Looking at the block diagram from cypress, overall device utilizes SECI UART and GCI GPIO, I presume the current BT driver utilizes the BT_HOST_WAKE via port control, can you confirm this to be true? 

I've also found a Broadcom SDIO driver that looks as though it work with ARM32 not ARM64 but cannot find source code for the 32bit driver to recompile to 64bit.

https://www.catalog.update.microsoft.com/Search.aspx?q=broadcom%20802.11n

 the one found was:

Broadcom - Other hardware - Broadcom 802.11n Wireless SDIO Adapter | Windows 10 and later drivers,Windows 10 and Later Servicing Drivers for testing | Drivers (Networking) | 7/25/2016 | 1.596.12.1 | 1.2 MB

[image]

worproject, 2020-12-19:
SDHost is no longer a viable option on the Pi 4 as it can't be routed to the SD card slot. We need EMMC2 drivers to free up the Arasan SDIO controller, which in turn will enumerate the WLAN adapter.

As you have discovered, the Wi-Fi driver is closed source and Cypress/Broadcom did not provide any ARM64 binary. We may have to write our own driver as we did for Bluetooth.

`I presume the current BT driver utilizes the BT_HOST_WAKE via port control, can you confirm this to be true?`
The wake pins are not exposed nor required. We do have access to WL_REG_ON and BT_REG_ON, but these are already configured by the firmware blob.

aiden1989acw, 2020-12-21:
Probably a stupid question but could we not take a look at linux driver as a solution to our issue?
I've found a link that should help with regards to an EMMC2 driver but obvious porting to windows would be necessary.

https://github.com/rsta2/circle/blob/master/addon/SDCard/emmc.cpp

worproject, 2020-12-21:
Yes, we can have a look at the Linux driver.
Currently the UEFI team is working on merging the EMMC2 changes to the upstream edk2-platforms repository. From what I've heard, the controller mostly complies with the standard SDHCI spec, but it requires some quirks.

Anonymus172017, 2021-04-02:
I'm sorry, I'm a rookie. But if you change the entrance of the LAN by the one of the wifi. I'm sorry, I'm spanish so my english is dreadful.

aiden1989acw, 2021-08-04:
Further to this conversation, I've install the Arasan driver under Windows 11 on arm (I'm booting the OS straight off of USB), the driver installation exposes the following Hardware IDs: "SD\VID_02d0&PID_a9a6", "&FN_1", "&FN_2", "&FN_3" to  device manager.

I have emailed Guangzhou Chenle Information Technology Company Limited, who have drivers on the Microsoft catalogue, presumably they have originally compiled versions for X86 and X64 so hoping for them to email back to recompile for ARM and ARM64.

danielgindi, 2021-09-24:
@aiden1989acw any news about the arm64 version? Did they reply?

zap8600, 2022-01-07:
Is there any news on the ARM64 drivers?

aiden1989acw, 2022-01-08:
> Is there any news on the ARM64 drivers?

Zero response from the driver developers I'm afraid. I think the consensus from most of the senior devs of wor, is that the Broadcom driver is closed source and they would need to stump up considerable amount of capital to ask Cypress to develop a 64bit wifi driver, and no one wants to take the task at hand to develop it from scratch unfortunately.

zap8600, 2022-01-08:
Thanks. If I could, I would, but I really can't. I don't know how to make
the drivers, or what libraries I need, or commands, or anything like that.
And like earlier, we would need the emmc2 driver finished to free SDIO.
Same thing for the GPU because I want hardware acceleration in my programs,
but same thing (well, except for SDIO). But thanks anyways.

CE1CECL, 2022-01-15:
I am afraid that even if we did have the 64-bit driver, we would face another problem: blobs! they never worked on the 3B+ so it wont work on the 4(00) either since they both use the same card(CYW43455/BCM43455). While @mariobalanica said in the devEco discord "it **might** work if you replace the firmware that comes with the driver and the config", if that doesn't work, then we would need a complete rewrite for it to work anyways. The 3B shouldn't be as big of a problem if we do get tho.

derryktutt, 2022-02-02:
this might help!
https://www.infineon.com/cms/en/product/wireless-connectivity/airoc-wi-fi-plus-bluetooth-combos/cyw43455/

derryktutt, 2022-03-02:
Ok, I might have found a solution! 
https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/building-arm64-drivers

CE1CECL, 2022-03-02:
> Ok, I might have found a solution! https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/building-arm64-drivers

Are you kidding me?
That is a solution on how to build a driver with source code for it, but there isn't any source code available for the RPi Wi-Fi
and for the link above, I have 0 clue on why you send a Info Website on the card, but that doesn't do any good for us, does it?

Not trying to be rude, just saying.

gitmanbloggus, 2023-05-15:
It's basically impossible to extract the drivers and redirect them to windows, right?
Those drivers are linux, so can we not port them over... at all?

CE1CECL, 2024-11-29:
FYI the Wi-Fi Driver ID (If it was exposed) should be: `SD\VID_02d0&PID_a9bf` Because `https://github.com/torvalds/linux/blob/7af08b57bcb9ebf78675c50069c54125c0a8b795/include/linux/mmc/sdio_ids.h`:
```
#define SDIO_VENDOR_ID_BROADCOM			0x02d0
#define SDIO_DEVICE_ID_BROADCOM_43455		0xa9bf
```
Correct me if I am wrong though, but there are X86 drivers on MS update (that wont work) here: https://www.catalog.update.microsoft.com/Search.aspx?q=SD%5CVID_02d0%26PID_a9bf

CE1CECL, 2024-11-29:
> a9a6

Which device is this? This seems like the `SDIO_DEVICE_ID_BROADCOM_43430`