worproject/RPi-Windows-Drivers #11: SD Controller setting to prevent BSoD on first boot (when Windows is installed on the SD card)

GNU-Linuxer, 2021-03-26:
When user finished flashing SD card, upon first boot, user need to immediately enter UEFI setting, then go to RPi configuration, select "MAX" under CPU (to speed up set-up process), and select "Arasan SDHCI controller".

Then user should press yes to save changes, and on main menu, select reset option to re-power the Pi. Also, I would recommend user to unplug keyboard (and as much USB device as possible) until it's ready for interactive setup.

After enter to Windows Desktop, the user can visit UEFI setting to switch back to eMMC controller for a faster performance, and USB device can always stay plugged in during boot without any issue.

I suggest to bring this notice evident in the imager program, as it prevents BSoD on first boot and subsequent "Windows Setup could not configure..." error (which requires re-flash the SD card).

Windows version is 19041.508

mariobalanica, 2021-03-26:
Did you choose GPT as the partition scheme?

GNU-Linuxer, 2021-03-26:
Hello, I choose MBR partition scheme.

mariobalanica, 2021-03-26:
What SD card are you using? Also the last log file from the imager would be useful.

GNU-Linuxer, 2021-03-27:
Sandisk Extreme Pro (A2 speed class) 64GB, and unfortunately the log is deleted.

CE1CECL, 2021-04-04:
might be related to https://github.com/raspberrypi/windows-drivers/issues/24

mariobalanica, 2021-04-04:
Could be, but we don't know what kind of BSoD the author of this issue is getting.
I'm going to close this issue and continue the discussion in the upstream repository: https://github.com/raspberrypi/windows-drivers/issues/24

closed 2021-04-04 by mariobalanica