worproject/RPi-Windows-Drivers #15: Audio Jack Issues

anonymousaga, 2021-05-02:
I realized that there were drivers for the audio jack today! However, the left and right channels are swapped. This is a minor inconvenience. However, especially with headphones, the major inconvenience is the loud pop when sound starts/ends and a high-pitched whine when the sound is playing. Is this an issue with the PWM?

mariobalanica, 2021-05-04:
Yes, the pops and background noise are due to the PWM audio hardware. A fix for the pop would be to never turn off the PWM channels, but this will make the hiss never stop.

As for the swapped channels, will be fixed in a future update.

anonymousaga, 2021-05-04:
Ok. In that case, I will use wired speakers instead of headphones when tinkering with the Pi. Also, I noticed that lower-quality headphones with less high-pitched capacity do not produce the hiss. I will close this issue when the audio channels are corrected. Thanks for your help.

anonymousaga, 2021-05-11:
By the way, could you send a message once the fixed driver comes out? This way, I will get an email notification and will upgrade my Pi's drivers quicker. Thanks again.

mariobalanica, 2021-05-12:
Sure.

mariobalanica, 2021-11-30:
@anonymousaga The audio pops have been fixed in the latest driver package.
I was going to say that the channels are not swapped, but I've just tested my headphones on another computer to be sure, and turns out they're swapped too. So that will be fixed in the next update.