worproject/RPi-Windows-Drivers #16: KERNEL_MODE_HEAP_CORRUPTION BSOD when using Speedtest.net

dennisameling, 2021-05-17:
Specs:
- Raspberry Pi 4 8GB
- 128GB SanDisk Extreme card (UHS-3)
- Windows 10 build 19041 (21H1)

Steps to reproduce:
1. Install the Speedtest.net app from the Microsoft Store: https://www.microsoft.com/en-us/p/speedtest-by-ookla/9nblggh4z1jc or use the website: https://speedtest.net
2. Start a test
3. The download test should finish correctly
4. When the upload test starts, you'll most likely get a KERNEL_MODE_HEAP_CORRUPTION BSOD

What I've tried already:
- Re-flashing doesn't help
- Switched to an external HDD instead of the SD card. The problem doesn't occur there, so I assume this has something to do with the SD card driver

Unfortunately, the BSOD appears for <1 second, then the device tries to reboot but fails at "failed to open device: 'sdcard'". When I unplug the power and try again, Windows boots normally again.

Please let me know if I can provide any logs/etc.

mariobalanica, 2021-05-27:
Can you try installing the `bcm2836sdhc` driver from https://github.com/worproject/RPi-Windows-Drivers/releases/tag/v0.6 and setting the SD controller to Arasan in UEFI settings?

dennisameling, 2021-05-31:
@mariobalanica thanks for your reply. Just re-flashed with the 0.6 drivers and set the SD controller to Arasan in the UEFI settings. Unfortunately, the BSOD still shows up with the steps mentioned above. The good thing is that the Pi actually reboots itself now after the BSOD, rather than showing `"failed to open device: 'sdcard'"` which I saw with the other driver.

Device Manager correctly reports that the Arasan driver is being used:

[image]

Anything else you'd like me to test/verify? This is not urgent by any means as I'm using an SSD now which works flawlessly. Just trying to help making the platform more stable 👍🏼 

Thanks in advance!

mariobalanica, 2021-06-02:
I was not able to reproduce this issue on 19041. Haven't tested 19043 yet.