worproject/RPi-Windows-Drivers #17: Dual monitor support with basic driver

stormenergy91, 2021-05-21:
Hi, there is any way to enable dual screen support also if actualy there is no working driver video?

For helping in build the driver video there are some resource (or documentation) about the work to do and the actual stete of the driver?

Rayman0710, 2021-05-22:
The team of the WoR project as already create a dual display driver but for the moment, it is in test by the team but if you want try it the team can give you the driver at your own risk on the discord server by private message

stormenergy91, 2021-05-22:
Thanks for this update. How i can ask for the experimental driver to discord?

mariobalanica, 2021-05-27:
You can find a link to the Discord server here: https://worproject.ml/community

stormenergy91, 2021-06-11:
If anyone is interested: https://github.com/themindvirus/pi4bdd

closed 2021-06-11 by stormenergy91