worproject/RPi-Windows-Drivers #18: Bluetooth Not working

Ddongddi, 2021-05-29:
Hello.
Do you know why is Bluetooth driver not working?
I installed the driver and it's giving me Code 10 error

This is the error : 
```This device cannot start. (Code 10)

{Drive Not Ready}
The drive is not ready for use; its door may be open. Please check drive %hs and make sure that a disk is inserted and that the drive door is closed.```

mariobalanica, 2021-07-03:
Fixed in the latest driver package release.

closed 2021-07-03 by mariobalanica

SquallATF, 2021-07-04:
I have updated SerPL011 and cywbtserialbus drivers to the latest version, but still get code 10 error?

mariobalanica, 2021-07-04:
Try re-installing the OS with the latest driver package. 
I cannot reproduce this issue on a Pi 4 running build 22000.1.