worproject/RPi-Windows-Drivers #19: [Question] WSL2 is supported?

stormenergy91, 2021-06-11:
Hi, I don't know if is a problem connected with the UEFI driver, but when I try to install WSL2 on Windows on my Pi 4, after enabled `Virtual Machine Platform` the system not start anymore. 
Anyone knows if there is a workaround, since with the next windows 10 release wsl2 is the default solution.
Thanks

gjsman, 2021-06-29:
@stormenergy91 Never going to happen. The Pi doesn't have Virtualization extensions.

mariobalanica, 2021-06-30:
This is a known issue. It's not driver-related, so I suggest reporting it here: https://github.com/pftf/RPi4/issues.

> Never going to happen. The Pi doesn't have Virtualization extensions.

@gjsman The Pi does support virtualization. See: [ESXi Arm](https://flings.vmware.com/esxi-arm-edition).

closed 2021-07-03 by mariobalanica

JeromTWinL, 2021-09-22:
@stormenergy91 the wsl2 was tested (by me) to work in windows 10 Raspberry Pi 3.