worproject/RPi-Windows-Drivers #25: gpu

cqy20091031mc, 2021-09-03:
What is the current Raspberry Pi video card driver process? I heard that your team has a beta gpu driver? Only 1080p? Why can't I use it?

JeromTWinL, 2021-11-05:
I think you're said about the roskmd.sys, btw the driver loads, but it doesn't work, the driver folder doesn't have the gpu driver.

cqy20091031mc, 2021-11-12:
But in Baidu, someone has successfully used the gpu driver. The driver is under test and does not support hardware acceleration, only 1080p.

ethpsa09, 2022-08-07:
I just found what cqy was talking about, https://www.bilibili.com/read/mobile?id=12900980, you will have to translate it to english. Possibly someone could look at the code and develop it more? I will test this driver and let you know of the results.

ethpsa09, 2022-08-07:
the only problem is that it looks like it only supports 1080p

ethpsa09, 2022-08-07:
and tvs with a higher/lower res show a black screen

ethpsa09, 2022-08-07:
here is my google drive mirror btw baidu downloads for overseas was so stupid so im going to save anyone who sees this the hassle: https://drive.google.com/file/d/1FwcXt916MJbHR-FzgeZJ-6uL0rOqUdEO/view?usp=drivesdk

ethpsa09, 2022-08-15:
wait nvm its the rokmnd .sys i think

binarymaster, 2023-07-20:
Please close this issue, it's duplicate of #3.