worproject/RPi-Windows-Drivers #29: Wifi driver issue, does it apply to external dongle?

PublicVoidUpdate, 2021-11-16:
With the wifi driver issues on WOR does this also apply to external wlan dongles? Or only internal?

SinclairSchneider, 2021-11-16:
So far I did not find any wifi dongle that comes with an Windows ARM64 driver. There are also no preinstalled drivers that come with the operation system. [I asked the same question in the Microsoft Community](https://answers.microsoft.com/en-us/windows/forum/all/usb-wifi-driver-for-windows-10-on-arm64/97ed88f7-296d-4a9b-84fd-3f0fd8e396dd) but so far without luck. If you find a solution please let me know. Also the wifi drivers for WindowsIoT can't be used since they are all 32 bit.

PublicVoidUpdate, 2021-11-16:
Damn, that suck. if I find any adapters with arm64 drivers I'll hit you guys up.

PublicVoidUpdate, 2021-11-18:
UPDATE, I have found a short term solution, if you pair your rpi with and Android/windows device with either mobile data or a wifi connection you can tether over Bluetooth. It's not by any means fast or efficient but it can work as a temporary solution.

lolimjustherefornoreason, 2024-01-23:
i know a wifi adapter that works here's the link for it (for the raspberrypi4) i have not tested it but i found it in a leepsp video

https://www.amazon.com/gp/product/B09NN1L4SP/ref=ox_sc_act_image_1?smid=A2SYB8XSOUVOA1&psc=1

lolimjustherefornoreason, 2024-01-23:
and the link of the video https://www.youtube.com/watch?v=LUoGqtq-u90&ab_channel=leepspvideo

lolimjustherefornoreason, 2024-01-23:
(there is some steps to configure the adapter so watch that video)