worproject/RPi-Windows-Drivers #30: Which driver is for which device ID in windows device manager?

jayjupdhig, 2021-11-22:
...instead of trying every combination!?

-> https://github.com/worproject/RPi-Windows-Drivers/releases

[image]

binarymaster, 2021-11-22:
`BCM2850` is Video Core 4 GPU

For more see: https://github.com/tianocore/edk2-platforms/blob/194269223294bb30513837d3582a6a635b672661/Platform/RaspberryPi/AcpiTables/GpuDevs.asl#L39

kosuodhmwa, 2021-11-22:
Thank you very much. But i don't just me that device. So you see, there are 5 unknown devices on my RPi 4 with Windows 11 Pro.

There is no list which device need what driver?

binarymaster, 2021-11-22:
I don't really know whether Windows ACPI supports displaying something as a device friendly name, when the driver is not installed for it yet.

kosuodhmwa, 2021-11-22:
So try-and-error - trying every combination? :-)

kosuodhmwa, 2021-11-30:
???

mariobalanica, 2021-11-30:
There are no drivers for those "Unknown devices".

kosuodhmwa, 2021-12-01:
ok thx

kosuodhmwa, 2021-12-01:
But for what are the drivers on https://github.com/worproject/RPi-Windows-Drivers/releases ?

mariobalanica, 2021-12-01:
https://github.com/worproject/RPi-Windows-Drivers#status

kosuodhmwa, 2021-12-01:
ok thx

closed 2021-12-21 by jayjupdhig