worproject/RPi-Windows-Drivers #32: Internet not working.

zap8600, 2022-01-07:
Hi. The ethernet drivers are not working for Windows 10 1904.1415. Are there any patches for it? Or a WiFi driver I could test? I'm getting tired of constantly having to hook up my phone for USB Tethering just to get internet.

mariobalanica, 2022-01-09:
Ethernet works fine on 19041 and newer.

zap8600, 2022-01-09:
Oh. I meant 19044.1415.

mariobalanica, 2022-01-09:
Do you see the network adapter in Device Manager? Does it have an yellow mark?

If not, it's likely an issue with your network configuration.

Strandizer, 2022-01-10:
I have similar issues with no network in 19044.1415 (UID: 09bff971-81f5-4cd4-b834-fc0e9ee83d59). The Yellow warning shows up in device manager. Message "Device cannot start. Code 10", "Object not found".
Used RPI4 oem3.inf driver from february on RPI400, bcmgenet can´t start.

zap8600, 2022-01-10:
There is no warning sign. How do I change my network configuration, and what do I need to change it to?

mariobalanica, 2022-01-11:
I've just installed 19044.1415 on a Pi 4 and Ethernet worked fine out-of-the-box. 

> There is no warning sign. How do I change my network configuration, and what do I need to change it to?

@zap8600 Check your cable and router settings.

> I have similar issues with no network in 19044.1415 (UID: 09bff971-81f5-4cd4-b834-fc0e9ee83d59). The Yellow warning shows up in device manager. Message "Device cannot start. Code 10", "Object not found".
Used RPI4 oem3.inf driver from february on RPI400, bcmgenet can´t start.

It'll probably be a bit difficult to track this down as I don't have a Pi 400 and it doesn't seem to be a widespread issue, so I'll need help with testing a few driver builds.

Can you try the driver below?
[bcmgenet_test1.zip](https://github.com/worproject/RPi-Windows-Drivers/files/7848490/bcmgenet_test1.zip)

zap8600, 2022-01-11:
I'll try to just hook up my ethernet cable. If it doesn't work, I'll try changing the settings. Or try that driver.

Strandizer, 2022-01-12:
Thx @zap8600 , The bcmgenet_test1 driver solved the network issue. Network is now fully working. Many thanks!
Out of curiosity what was updated to the driver?

zap8600, 2022-01-12:
I never sent the driver.

Strandizer, 2022-01-12:
Ohh, Did @mariobalanica upload the driver. I misinterpreted the respons above.
Anyhow. The issue is resolved.

mariobalanica, 2022-01-12:
@Strandizer Cool, can you send me some logs from the driver using the Traceview tool below?
[traceview.zip](https://github.com/worproject/RPi-Windows-Drivers/files/7855832/traceview.zip)

Instructions: 

1. Uninstall the GENET Ethernet driver.
2. Extract the archive, then open `traceview.exe` as administrator on your Pi.
3. Click on _File_ -> _Create new log session_.
4. Select the `bcmgenet_netadaptercx20.pdb` file from the archive and click on _Next_.
5. Select _Log trace event data to file_ and save the file to an easily accessible location (like Desktop), then click on _Finish_.
6. Install the GENET driver from `bcmgenet_test1.zip` and wait until you see some logs appear in the Traceview tool.
7. Close the tool and upload the generated .etl file here.

mariobalanica, 2022-01-18:
Fixed in the latest driver package release.

closed 2022-01-18 by mariobalanica