worproject/RPi-Windows-Drivers #33: Ethernet issues

zap8600, 2022-01-18:
I'm still having trouble with the Ethernet. My RPi4 bluescreened trying to find updates. This didn't happen when I was using a Ethernet to USB (or the other way around), so I don't know what to do. I'm using a brand new install. What do I do?

zap8600, 2022-01-25:
I'm still having Ethernet issues

mariobalanica, 2022-01-26:
What build of Windows are you using? And what's the stop code in the BSoD?

zap8600, 2022-01-26:
I'm using version 19044.1466. I think the stop code was SYSTEM_PTE or something similar.

zap8600, 2022-01-29:
I got a new "green-screen" with a new stop code. 
PAGE_FAULT_IN_NONPAGED_AREA

zap8600, 2022-01-29:
For some reason, the Ethernet now works on my Pi. I just can't have it plugged in on boot, or it crashes.

zap8600, 2022-01-31:
Never mind. I'm still having problems. I've started to do all of my important work on my Pi. When I was about to test code, it bluescreened. My code was fine, but any builds were corrupted, so I had to restart. And no, I didn't get the stop code that time. It just flashed the screen, and then rebooted.