worproject/RPi-Windows-Drivers #38: Info: About Drivers

Corsair-63, 2022-04-22:
First of all, you must be very proud of the work and efforts you are doing, I am very happy because I was able to configure my Rpi with Windows 11, and most of the devices work, also installed with the software from https://www.worproject.ml/ on an SSD giving much more speed and reliability to the system.

I've these devices not yet configured in the device manager:
- ACPI\VEN_BCM&DEV_2850
- ACPI\VEN_BCM&DEV_2854
- ACPI\VEN_PNP&DEV_0C06
- ACPI\VEN_BCM&DEV_2835
- ACPI\VEN_BCM&DEV_2847
- ACPI\VEN_BCM&DEV_2848

Also, I have already seen in these threads that there is a problem with the audio jack driver and that it produces some noise while the jack is connected. The same thing happens to me making it impossible to have the speakers connected by Aux plug.

I have two questions:

1 - Those devices which is who?, I know that one is the Wi-Fi, but I have no idea what it is and the others, no idea what they are?
2 - Could the drivers be updated manually? instead of doing the process in the WOR FAQ?

My machine is the following:
Procesador	BCM2711 (ARM Cortex-A72)   1.50 GHz
RAM instalada	8,00 GB (7,85 GB usable)

Tipo de sistema	Sistema operativo de 64 bits, procesador basado en ARM
Lápiz y entrada táctil	La entrada táctil o manuscrita no está disponible para esta pantalla

Edición	Windows 11 Pro
Versión	21H2
Instalado el	‎13/‎04/‎2022
Versión del sistema operativo	22000.613
Experiencia	Paquete de experiencia de características de Windows 1000.22000.613.0

The windows version is the latest stable and also registered.

mariobalanica, 2022-04-28:
> 1 - Those devices which is who?, I know that one is the Wi-Fi, but I have no idea what it is and the others, no idea what they are?

You can find their definitions in the ACPI tables: https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/AcpiTables
There are no drivers for those devices.

> 2 - Could the drivers be updated manually? instead of doing the process in the WOR FAQ?

The process described on the FAQ page (https://worproject.ml/faq#how-can-i-update-the-drivers) shows you how to update the drivers manually.

binarymaster, 2022-04-28:
> I know that one is the Wi-Fi

Actually none of the devices you have listed is the Wi-Fi. :slightly_smiling_face: 

The closest one is `ACPI\VEN_BCM&DEV_2847` which is Arasan SD Host Controller - the Wi-Fi thing is actually attached to it via SDIO. So if you want the Wi-Fi thing, you'll have to get an SD bus working first.