worproject/RPi-Windows-Drivers #40: Question: How to update driver clarification?

Corsair-63, 2022-05-02:
Hi,

just working around on how to update driver?

as per WoR FAQ, the safest way of doing it, could be:

> Run Dism /Image:E: /Add-Driver /Driver:"PathToDrivers" /Recurse /ForceUnsigned
Replace PathToDrivers with the path to the extracted driver package.
Replace E: with the letter of the Windows partition on your Raspberry Pi boot drive.

as usual the Boot partition uses to have no letter assigned, so, How to proceed? with no disk letter known?

could be valid option to run? Dism /Image:BOOT: /Add-Driver /Driver:"PathToDrivers" /Recurse /ForceUnsigned

or should I to name the BOOT partition with a drive letter (AFAIK, A & B are not allowed and of course anyone in use in the computer) also would be any problem for the system that BOOT partition would be named with a drive letter after?

naming I should have to use the DISKPART option?,  as:

- list disk
- sel disk
- list vol (check the vol for BOOT)
- sel vol X ( ... the info of last command)
- assign letter H (any available letter)
- exit
- dir /a /b /s H: (or the named drive you did it)
- Dism /Image:H: /Add-Driver /Driver:"PathToDrivers" /Recurse /ForceUnsigned

because I did the second option of the WoR FAQ: 
pnputil.exe /add-driver "D:\Drivers\*.inf" /subdirs /install /reboot 
everything was fine and 4 drivers were added.
after reboot the RasPi 4 was not able to boot because the HD was not bootable ??????
again I had to do all procedure of creating SSD latest with Imager and giving the option of download latest UEFI and Drivers 0.16 from here.

all evening and early morning lost 😭 to setup everything again 😄

mariobalanica, 2022-05-05:
> as usual the Boot partition uses to have no letter assigned, so, How to proceed? with no disk letter known?

Drivers are installed on the Windows partition, there's no need to touch the Boot one.
It should already have a letter assigned, but if it doesn't, you can do it via Disk Management or diskpart. 

> after reboot the RasPi 4 was not able to boot because the HD was not bootable ??????
Are you booting from USB or an SD card?

Corsair-63, 2022-05-05:
> Drivers are installed on the Windows partition, there's no need to touch the Boot one. It should already have a letter assigned, but if it doesn't, you can do it via Disk Management or diskpart.

that means in most of cases should be C: (where the OS (Windows) is installed)??

> > after reboot the RasPi 4 was not able to boot because the HD was not bootable ??????
> 
> Are you booting from USB or an SD card?

I was booting from SSD on USB same that I was booting and working before proceeding with the second procedure:
pnputil.exe /add-driver "D:\Drivers\*.inf" /subdirs /install /reboot
and rebooted the system for update de results of above

mariobalanica, 2022-05-05:
> that means in most of cases should be C: (where the OS (Windows) is installed)??

I'm talking about the Windows partition of your RPi installation. C:\ is by default the Windows partition on your PC, you don't want to touch that.

> I was booting from SSD on USB same that I was booting and working before proceeding with the second procedure:

Are you using Windows 10 on the Pi? If so, this is a known issue with driver package v0.16.

Corsair-63, 2022-05-08:
Yesterday updated to version 0.17 with the procedure:

> Dism /Image:BOOT: /Add-Driver /Driver:"PathToDrivers" /Recurse /ForceUnsigned

in my main desktop and everything was really fine also now booting is faster than before.

I was surprised when connecting the Raspi SSD in the USB enclosure to desktop and was assigned letter B:, old times reserved for floppies 🤣 also I unzip driver in D:\DRV so the correct string was:
Dism /Image:B: /Add-Driver /Driver:"D:\DRV" /Recurse /ForceUnsigned
no surprise and again SSD fitted to RasPi and now working fine.