worproject/RPi-Windows-Drivers #47: Is this project dead?

MinGW32-2006, 2023-05-27:
It's 2023 now. Is the project still under development or it's dead? The latest repository update is almost one year ago.

Mabbs, 2023-05-27:
rpi is dead, so the project is also dead

commandcontrolQ, 2023-06-06:
> rpi is dead, so the project is also dead

What makes you think that the Raspberry Pi is dead? Just curious.

Mabbs, 2023-06-07:
> > rpi is dead, so the project is also dead
> 
> What makes you think that the Raspberry Pi is dead? Just curious.

Because it hasn't been updated for a long time

MinGW32-2006, 2023-07-20:
So there is someone writing the GPU driver FROM SCRATCH? I'll wait for it. :)

binarymaster, 2023-07-20:
> So there is someone writing the GPU driver FROM SCRATCH?

GPU driver question is being tracked in #3.

MinGW32-2006, 2023-07-22:
> > So there is someone writing the GPU driver FROM SCRATCH?
> 
> GPU driver question is being tracked in #3.

I see. Thanks for you reply. :)

closed 2023-07-22 by MinGW32-2006

RetroSoftwareRepository, 2024-01-10:
> rpi is dead, so the project is also dead

It's not dead.... The Pi 5 is a thing now.