worproject/RPi-Windows-Drivers #48: ¿Como pruebo los pines GPIO con el windows11ARM?

darcknetss, 2023-06-21:
No logro encontrar la forma de como usarlo
en linux es el RPI.GPIO
E logrado ver para windows y solo me sale el Windows.Devices.Gpio
pero intentando solo me sale no defined

driver1998, 2023-06-26:
If you are talking about python you can try to use Windows.Devices.Gpio API with the pywinrt [winsdk](https://pypi.org/project/winsdk/) package.

darcknetss, 2023-06-26:
> Si está hablando de python, puede intentar usar la API de Windows.Devices.Gpio con el paquete pywinrt [winsdk](https://pypi.org/project/winsdk/) .
> 
> Si se siente cómodo con C++, puede consultar mi repositorio de demostraciones: https://github.com/driver1998/rpi3win10demos . Las demostraciones oficiales de .NET IoT también deberían funcionar.

darcknetss, 2023-06-26:
@driver1998 What I want to do more than anything is a .py that turns a led on and off.
well it is as an example
and another that reads pulsometer pulses
in ubuntu it used the RPi.GPIO

import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)

pin_led = 18

GPIO.setup(pin_led, GPIO.OUT)

GPIO.output(pin_led, GPIO.HIGH)
print("LED encendido")

GPIO.output(pin_led, GPIO.LOW)
print("LED apagado")

GPIO.cleanup()